E-racingBergamo/ECU-Issues #7: Luce Freno

Smolsolli, 2025-04-22:
- [ ] definire logica che accenda luce del freno quando viene premuto pedale del freno

nonfederico, 2025-04-22:
Bozza circuito fatta su ltspice, da test funziona.

TODO:
Fare pcb

nonfederico, 2025-04-24:
problema con selezione e reperimento dei componenti da usare nella scheda, non esistono modelli spice utilizzabili durante la simulazione. 
devo posticipare la scadenza per la creazione della scheda

nonfederico, 2025-04-24:
problema con selezione e reperimento dei componenti da usare nella scheda, quelli precedenti non rispettano le caratteristiche e non esistono modelli spice dei nuovi componenti utilizzabili durante la simulazione.
devo posticipare la scadenza per la creazione della scheda

Smolsolli, 2025-05-21:
Relè 1: https://it.rs-online.com/web/p/alimentazione-motori-e-strumenti-di-sviluppo-per-robotica/8430834?cm_mmc=IT-PLA-DS3A-_-google-_-CSS_IT_IT_PMAX_Catch+All_NEW-_--_-8430834&matchtype=&&gad_source=1&gad_campaignid=21013128907&gbraid=0AAAAADkeWNMEUpdWYEsEa2e6hqxtDcHNA&gclid=Cj0KCQjwxJvBBhDuARIsAGUgNfhWUqR_x5-D-gdIVXu5sC2-WaMCu6Ydo55_6AU3pv6I4UexmF58h5kaAkfvEALw_wcB&gclsrc=aw.ds
Relè 2 : https://www.digikey.it/it/products/detail/sunfounder/TS0010D/18668618?gad_source=1&gad_campaignid=20087055286&gbraid=0AAAAADrbLljpg8OO4XpHfwn_35h96BAPn&gclid=Cj0KCQjwxJvBBhDuARIsAGUgNfhaqBmtxUpJH7d454ignfkx7__gvMGKccpc5cAa3ksuiwl2lnHYvLoaAivYEALw_wcB&gclsrc=aw.ds

s-stasi, 2025-09-17:
Fatto