E-racingBergamo/ECU-Issues #8: Design cockpit

Smolsolli, 2025-04-22:
- [ ] effettuare verifiche dimensionali 
- [ ] Definire Attacchi per Led e Bottoni (Attenzione a regolamento)
- [ ] Disegnare su CAD il cockpit completo

RM-Mino, 2025-04-23:
E' stata completata l'analisi del regolamento, al fine di individuare quanti e quali -tra pulsanti, led e swith- è necessario avere.
E' emerso che dovranno quindi esserci:
- 3 led: 
          - 1 Verde (Che indica quando il TS è spento)
          - 2 Rossi (La prima che si accende se l'AMS ha aperto l'SDC, la seconda che si accende se l'IMD ha aperto l'SDC)
- 1 pulsante:
          - Shut-Down Button (Pulsante rosso dal diametro minimo di 24mm)

RM-Mino, 2025-04-24:
Sono stati scelti i led e il pulsante da acquistare.
Oltre a ciò è stato deciso di inserire un ulteriore led che indica il corretto funzionamento della centralina durante la marcia del veicolo.

Di seguito vengono elencati i link che riconducono ai pezzi da acquistare:

SHUT-DOWN Button:
https://www.digikey.it/it/products/detail/apem-inc/A01ESDM/6187656?gclsrc=aw.ds&&utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=Pmax%20Shopping_Supplier_Automation%20and%20Control&utm_term=&utm_content=&utm_id=go_cmp-21267066609_adg-_ad-__dev-c_ext-_prd-6187656_sig-CjwKCAjwwqfABhBcEiwAZJjC3rqrG6SxWRBxlmrloO-bFVhA0v9Gi7YZbZem5nxcHQOtnuMArWeaMBoCaYMQAvD_BwE&gad_source=1&gbraid=0AAAAADrbLlgqKeAKl4U5xO5SIokpg-0NK&gclid=CjwKCAjwwqfABhBcEiwAZJjC3rqrG6SxWRBxlmrloO-bFVhA0v9Gi7YZbZem5nxcHQOtnuMArWeaMBoCaYMQAvD_BwE&gclsrc=aw.ds

RED Led (x2):
https://it.rs-online.com/web/p/led/2285821?gb=s

GREEN Led (x1):
https://it.rs-online.com/web/p/led/2286004?gb=s

BLUE Led (x1):
https://it.rs-online.com/web/p/led/4663532?gb=s

RM-Mino, 2025-04-28:
Sono stati creati i fori in cui andranno inseriti i led (4 totali: 2 rossi, 1 verde, 1 blu) e il pulsante di Shut-Down (conforme alle richieste regolamentari -> diametro pulsante di 24mm)

Si ricorda che -una volta che si avrà a disposizione la piastra- sotto ogni led e pulsante, andrà posta un etichetta indicante a cosa si riferisce ciascuno di essi

[image]

RM-Mino, 2025-04-28:
> * [x]  Controllo del regolamento[x]  Verifica della precedente lista di Pulsanti e Led[x]  Selezione componenti da acquistare[x]  Effettuare verifiche dimensionali[ ]  Definire Attacchi per Led e Bottoni (Attenzione a regolamento)[x]  Disegnare su CAD i fori presenti sulla piastra

RM-Mino, 2025-04-29:
Sono stati modificati i led che si intende acquistare e posizionare sulla plancia alle spalle del volante
(I precedenti avevano una tensione nominale insufficiente, ed erano quindi inadatti). 

Di seguito vengono elencati i link che conducono ai nuovi componenti

SHUT-DOWN Button:
https://catalogue.lovatoelectric.com/it_it/Operatore-pulsante-a-fungo-serie-Platinum-metallica-%C3%9822mm-ad-aggancio-sgancio-a-rotazione-%C3%9830mm-Per-arresto-d-emergenza-conforme-ISO-13850-Rosso/LPSB6634/snp

RED Led (x2):
https://it.rs-online.com/web/p/indicatori-da-pannello/7227448?gb=s

GREEN Led (x1):
https://it.rs-online.com/web/p/indicatori-da-pannello/7227441?gb=s

BLUE Led (x1):
https://it.rs-online.com/web/p/indicatori-da-pannello/0203846?gb=s

(Anche il pulsante di Shut-Down è stato sostituito, in queto caso con un modello fornito da Lovato)

--> In virtù dei cambiamenti fatti, sono stati modificati anche i diametri dei fori fatti sulla plancia

RM-Mino, 2025-04-29:
E' stato aggiunto -al 3D della plancia- anche il modello del pulsante di Shut-Down, e sono state quindi fatte le opportune verifiche dimensionali, per verificare la correttezza del foro.

Di seguito si mostra lo stato attuale del 3D:

RM-Mino, 2025-05-12:
Sono stati nuovamente sostituiti i led, andando su una soluzione più economica, ma ugualmente efficiente.
I nuovi led -di cui si allega il link di seguito- hanno un corpo più piccolo, perciò non saranno d'intralcio alla scatola di controllo e possono essere alimentati sia a 12V che a 24V, e il diametro del foro di fissaggio è circa la metà.

Link: https://www.amazon.it/Gebildet-Precablato-Indicatore-Luminoso-Impermeabili/dp/B088D1M47R?pd_rd_w=jkw9i&content-id=amzn1.sym.bb1b8560-2899-4163-aee6-d19cc4760f97&pf_rd_p=bb1b8560-2899-4163-aee6-d19cc4760f97&pf_rd_r=GM6J2VG59B5YY2A6PFBB&pd_rd_wg=dJMWe&pd_rd_r=815a8ca1-e879-4354-890a-4170df3c588e&ref_=sspa_dk_detail_2&sp_csd=d2lkZ2V0TmFtZT1zcF9kZXRhaWxfdGhlbWF0aWM&th=1

Si è già provveduto a correggere il diametro dei fori fatti sul 3D della plancia.

NOTA: Si specifica che -almeno al momento- non si hanno però i 3D dei led citati

RM-Mino, 2025-05-13:
NON Necessario, ma per la presentazione di giovedì cercherò di avere un 3d completo (comprensivo dei led) da far vedere.
I led che inserirò NON saranno quelli effettivi, dato che di quelli non c'è il 3D, ma almeno si avrà un idea di come apparirà la plancia. (E' puramente a scopo illustrativo, giusto per non far vedere una piastra forata e basta)
Ad ogni modo il lavoro è terminato

[image]

closed 2025-05-13 by RM-Mino

RM-Mino, 2025-05-14:
E' stata realizzata una simulazione -PURAMENTE ILLUSTRATIVA- di come apparirà la plancia con i led ed il pulsante
(I led NON sono effettivamente quelli che andranno usati, ma sono più grandi di circa 10mm. Non sono stati inseriti direttamente quelli corretti data l'assenza di un 3D degli stessi)

[image]